PrintersForAnts/AntHead #11: 30mm hotend screws bottom out on all carriages

Foofinator94, 2024-11-22:
The update to 30mm length from 25mm doesn't work right now. The screws bottom out on both printed and CNC carriages.

Reccomend to make the clamped thickness increase by ~0.8mm to accomodate, or revert to the previous design with 25mm screws as that was working fine. 

[image]

[image]

[image]

hartk1213, 2024-12-16:
where did you see that it uses 30mm? those should still be 25mm

Foofinator94, 2024-12-16:
The previous iterations had more recessed counterbores for the screws, so needed 25mm. The latest didnt have enough thread engaging for 25mm so I thought 30mm was needed.

commit dbbcb995 (2025-02-06): various updates and fixed #11

* updated all hotend mounts to fix the mounting screw length to m3x25
* added support for Dragon Ace hotend
* added support for ReVolcano (Revo voron heatsink with hemera heatbreak and volcano heater)
* fixed rapido mount to use correct mounting m2.5 screws
closed 2025-02-06 by hartk1213