PrintersForAnts/AntHead #17: the bambu x1 hotend needs to shft to the side a little

broncosis, 2025-01-15:
when you put the hotend on its mount into the tool it forces the side of the cowl 
out a little bit I will take some pictures and post in a little bit 

granted I pulled it out of you hotend line up step file so I don't think its Released yet but I wanted to test I may try and pull it into fusion 
and make the slight adjustments just over a millimeter

broncosis, 2025-01-15:
[image]
[image]

broncosis, 2025-01-15:
I made some minor tweaks to hotend mount printing it tonight I will update 

@hartk1213  thank you for the work you have put in sofar if this work I will have to figure out git and put in a pull request

broncosis, 2025-01-16:
Yes a 1.1mm adjustment 
Was perfect 
[image]

tjeffoo, 2025-04-02:
Where can i get this hotend mount file? Its different from the one in the stl map. With the one in the stl map the x1c hotend does not fit with the stealthchanger backplate. This one is 90 degrees turned.

broncosis, 2025-04-04:
@tjeffoo  are you on the stealth changer Discord message me there I will send you my file to try out

marcoaccasarin, 2026-02-06:
I made a small adjustment to the main body to better fit the TZ 2.0. It worked well for me...

[image]

greguy43, 2026-02-14:
Hello, I am also looking for the STEP file for the Bambu X1C hotend; would it be possible to obtain it?