PrintersForAnts/AntHead #32: Extruder mounting holes

rbc1225, 2025-03-30:
I printed out both extruders under STLs and then Extruders and neither one of them line up with the mounting holes on the shell. I may be doing something wrong but I then printed out a Sherpa Mini extruder from another source it is spot on to be able be mounted to the shell. As you can see, I printed both bases from the extruder directory and both are spaced the same and don't line up with the shell.  I even tried your adapter plates thinking something would line up there. No luck. Anyway, not exactly sure how this works on Github but that's what I found. 

[image]
[image]

bcutty, 2025-03-30:
The two anthead extruders have been altered from the orbiter mounting pattern wwbmg and wwg2, so you’d need to print the orbiter mounting pattern cowl for the holes to line up instead of the sherpa mini one.

rbc1225, 2025-03-31:
> The two anthead extruders have been altered from the orbiter mounting pattern wwbmg and wwg2, so you’d need to print the orbiter mounting pattern cowl for the holes to line up instead of the sherpa mini one.

Ahhh Okay, I will look into that.  Thanks